dbisu/pico-ducky #262: Unable to enter setup mode

jw762, 2024-05-26:
Hello, having an issue with getting setup mode to work. I've followed the instructions to get everything properly installed several times and everything seems to work fine, including pin 15 for USB Storage, the other pin options for switching payloads, but I cannot seem to enter into Setup mode by using pin 1 (GP0). 

- When I do connect with GP0, Windows makes the sound as if it recognizes it, but nothing happens. No drive is shown to enter into. Even tried using a partition manager to see if it could recognize a drive, and nothing.
- When connecting pin 15 for USB Storage, the drive pops up, but it also executes payload.dd.
- And of course plugging the pico in with no pins connected just executes the first payload as usual.

So, no matter what I do, I cannot seem to edit the files without executing payloads. 
Any ideas here? Or is it possible I don't understand how it's supposed to work? 

Specs:

- This machine is running Windows 10 
- I'm using a Pico W
- Using pico-ducky-v2.0-us released on 2024-05-29
- Using adafruit-circuitpython-bundle-8.x-mpy-20240525

If you need anymore info let me know. Thank you.

dbisu, 2024-05-26:
On a Pico W, you also need to enable USB storage mode. By default a Pico W will not show up as a USB drive.
https://github.com/dbisu/pico-ducky?tab=readme-ov-file#usb-enabledisable-mode

jw762, 2024-05-27:
You're right, not sure why I didn't think about enabling both pins. Thank you for the quick response.

closed 2024-05-27 by jw762