dbisu/pico-ducky #263: Pico W Ducky - Backslashes converted to # using Neradoc UK Keyboard Layout

mick-dee-irl, 2024-06-03:
Hi - I have setup a Pico Ducky using Pico W. I have been able to get some test scripts to work. I am working on a simple use case where a text file is saved to a location on the target machine. I am using the UK Keyboard layout as provided by Neradoc. Below is my duckyscript that i am transferring to the target machine. However, the backslashes are being converted to # when entered onto the target machine via the Pico Ducky.

I am wondering if anyone has found a fix for backslashes on UK Keyboard layouts? 

DELAY 500
GUI r
DELAY 500
STRING powershell
ENTER
DELAY 500
STRING $filepath = "$env:userprofile\Documents"
ENTER
DELAY 500
STRING New-Item -Path "$filepath\secrets.txt" -ItemType File -Force
ENTER
DELAY 500
STRING $doc = "$filepath\secrets.txt"
ENTER
DELAY 500
STRING Set-Content -Path $doc -Value 'These are my secrets'
ENTER

io1901, 2024-09-08:
"#" is the "\\" key on the US layout.  You've  not done something silly like not replacing the lines in the duckinpython.py? I tested it and if you don't make the changes below the "\\" comes out as a "#".

```
# comment out these lines for non_US keyboards
#from adafruit_hid.keyboard_layout_us import KeyboardLayoutUS as KeyboardLayout
#from adafruit_hid.keycode import Keycode

# uncomment these lines for non_US keyboards
# replace LANG with appropriate language
from keyboard_layout_win_uk import KeyboardLayout
from keycode_win_uk import Keycode
```

dbisu, 2024-09-21:
Have you tried getting the latest libraries from Neradoc?
https://github.com/Neradoc/Circuitpython_Keyboard_Layouts/releases/tag/20231122

dbisu, 2024-12-21:
Try updating to the v3.0 release to see if that fixes this problem. There is a WIN_UK version now.