dbisu/pico-ducky #265: Payload not running

aloha9133, 2024-07-06:
So i tried to make a bad usb but when i plug the pi in my laptop which is windows 11 nothing happenes i followed exact steps and im sure i did nothing wrong although i bought the raspberry pi from aliexpress their are some extra files in it im not sure about i would really apprecieate some help

ibrahimxq, 2024-07-09:
same here please someone help us

aloha9133, 2024-07-10:
> same here please someone help us

hi Ibrahim so i figured what the issue is. 
i opened my old very old laptop and it had the old pico ducky files already installed from dbisu and when i tried those files the payloads worked so if u can get the older files your issue will be fixed

dbisu, 2024-07-11:
It would be helpful in debugging if you had logs from the debug serial port. Are you installing circuitpython 8.x or 9.x? I have not verified 9.x support yet.

ibrahimxq, 2024-07-11:
yeah, I figured it out and the problem was not reading. hahahaha

dbisu, 2024-09-21:
Closing, resolved.

closed 2024-09-21 by dbisu