dbisu/pico-ducky #269: No payload running, still showing up as a usb.

Retro7Development, 2024-07-27:
**Describe the issue**
When I plug in the pico, it doesn't run the payload, instead shows up as a usb. If I push the jumper cable harder and reconnect, nothing shows up, but still decides not to run.
**To Reproduce**
Use the RPI Pico as normally
**Expected behavior**
Nothing works as it should

**Additional context**
Using the default payload after installing in setup mode

dbisu, 2024-07-27:
Are you using a pico or pico w? connecting the jumpers is easier if you solder on pin headers to make the wire connections better.

Retro7Development, 2024-07-27:
Pico :)

dbisu, 2024-08-05:
I'll need information from the debug serial to help any further.
As for the jumper connections, my recommendation would be to solder on the header block so you can use wires with crimped ends to provide a more solid connection.

dbisu, 2024-09-21:
Ensure you are using CircuitPython 8.x and not 9.x.
Closing.

closed 2024-09-21 by dbisu