dbisu/pico-ducky #271: No payload running, still showing up as a usb.

sarimwaniR, 2024-08-04:
Describe the issue
When I plug in the pico, it doesn't run the payload, instead shows up as a usb. If I push the jumper cable harder and reconnect, nothing shows up, but still decides not to run.
To Reproduce
Use the RPI Pico as normally
Expected behavior
Nothing works as it should

Additional context
Using the default payload after installing in setup mode

(could not understand issue #269 little weak english please help)

dbisu, 2024-08-05:
For a Pico W, there are two jumper options. One controls if the pico shows up as a usb device, the other controls if the payload runs on boot.
Check both jumpers to makes sure they are in the expected states.
it might be easier to control if you solder the header block to the pins and then use a wire with correct ends to make the jumper.

sarimwaniR, 2024-08-05:
no I have the normal one

sarimwaniR, 2024-08-05:
here is the video from start
[](url)https://drive.google.com/file/d/1mXAWOBFQyRVAzJwXiDy58AjDAMp6Xyhg/view?usp=sharing

sarimwaniR, 2024-08-06:
please  help

sarimwaniR, 2024-08-07:
#273

closed 2024-08-07 by sarimwaniR

M1r0zo, 2024-08-12:
same problem. it used to work, now just it flashes 3 times, stops, flashes 2 times and then again 2 times in loop