dbisu/pico-ducky #272: Troubleshooting Pico Ducky

Anakinjh, 2024-08-05:
I watched the NetworkChuck video linked at the bottom of the page a few weeks ago, and i thought this project seemed incredibly cool. I ordered a pair of standard Pico boards because I was gonna make one for my brother, and I have been trying to get them setup for more than a week now.

My first question, which feels dumb but I'm really new to all of this is if the actual name of the device matters. I am doing all of my setup on Linux. running the command `lsusb` gave me a different name than "RPI-RP2"  but it was still very clearly my device so I continued. But I then have 2 issues regarding the next step, one of which being the name. Running `lsusb` now lists the device as "Adafruit Pico" but again, I didn't think anything of this at the time, but now I am reconsidering as all the instruction I have found online mention a "CIRCUITPY" device and I don't get that name on mine.

My other issue is that the device is intended to disconnect an reconnect automatically. On my Linux distrobution, out of the box, external storage devices don't mount on their own and I have chosen to keep it that way, but is this causing issues? I manually unmount and remount the device when setting up. 

I also had an issue where somehow the device became a read-only file system but I think that was because of something I did unknowingly but I thought I'd mention it. I ran `sudo shred /dev/sdx` after this issue and i followed all the steps in the guide again but still no luck.

I also typed up my own version of the guide I could follow so it was easier for me to follow in the future, with a little more detail and commands I could copy and run to ensure it was the same everytime, because my intention was to make a script that would do the setup for me once i got one working. My write up should be linked below as it will describe my exact steps I followed setting the device up the first, as well as each subsequent time 

Thank you for any advice!

[PicoDucky.txt](https://github.com/user-attachments/files/16500069/PicoDucky.txt)

dbisu, 2024-08-05:
On the first question, the only reason the name matters is so you know if it is in the bootloader or main app. RPI-RP2 is how the bootloader enumerates. This is where you can update the uf2. It mounts as CIRCUITPYTHON when running the normal app.
For the second question. the device showing up as a USB drive is only needed if you have files that your payloads use, or if you are updating the payloads.
Debugging your filesystem issues is out of scope for this project. there are too many variables since you are not running something standard.

dbisu, 2024-09-21:
Ensure you are using CircuitPython 8.x and not 9.x. Support has not yet been added for 9.x
Closing.

closed 2024-09-21 by dbisu