dbisu/pico-ducky #273: payload does not Start

sarimwaniR, 2024-08-07:
**Describe the issue**
I do everthing but the payload does not work (I have a normal pie)

**To Reproduce**
Steps to reproduce the behavior:
here is the video from start (ignore the sound)
https://drive.google.com/file/d/1mXAWOBFQyRVAzJwXiDy58AjDAMp6Xyhg/view?usp=sharing

**Expected behavior**
To work

**Debug info**
Nil

**Additional context**
please help

Aftab700, 2024-08-09:
I think you are facing this issue because of the different versions of `CircuitPython` and `Adafruit-circuitpython-bundle`.

You are using `CircuitPython 9.1.1` and `adafruit-circuitpython-bundle-8.x`.

Generally, CircuitPython X.* requires the X.x bundle. E.g., 8.2.9 required an 8.x bundle; 9.0.0 requires the 9.x bundle.

You can also try to debug this using PuTTY. Connect to the correct `COM` port using a serial connection type.

[image]

I had the same issue, and it was giving an error like below because of different versions:

```
code.py output:
Traceback (most recent call last):
  File "code.py", line 14, in <module>
  File "duckyinpython.py", line 9, in <module>
ValueError: incompatible .mpy file
```

[image]

Aftab700, 2024-08-09:
You can download this file: https://github.com/dbisu/pico-ducky/releases/download/v2.0/pico-ducky-v2.0-us.zip

or download the CircuitPython Version 8.x from here:
- [CircuitPython for the Raspberry Pi Pico.](https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/raspberry_pi_pico/en_US/)
- [CircuitPython for the Raspberry Pi Pico W.](https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/raspberry_pi_pico_w/en_US/)