dbisu/pico-ducky #276: Not working

Leapingcactus, 2024-08-14:
I used all of the files correctly and put them into the correct places, I followed a YouTube tutorial by Austin's lab, and then s tutorial bu network chuck, while following the tutorial by network chuck I deleted some files and cant get them back. Can I please have some help

dbisu, 2024-08-15:
Your best bet is probably to reset the device with [these](https://github.com/dbisu/pico-ducky/blob/main/RESET.md) instructions and then follow the install instructions [here](https://github.com/dbisu/pico-ducky/releases/tag/v2.0)

Leapingcactus, 2024-08-15:
Thanks I will let you know if it works, and are there any devices that it
will not work on?

dbisu, 2024-08-16:
This code has only been tested with RPi Pico and RPI Pico W boards. It might work with other similar boards that run CircuitPython.

Leapingcactus, 2024-08-16:
Thanks I did it and now it works perfectly, I was using the newest version
of circuit py instead of 8.0.0. Just a question, could I use this code on
picks and sell them?

dbisu, 2024-08-17:
You can, though keep in mind that the code is released as GPLv2 open-source, so you are really just selling the hardware. The code is released for free.

dbisu, 2024-09-08:
closing, resolved.

closed 2024-09-08 by dbisu