dbisu/pico-ducky #278: Scripts are not running, only opens pico file directory on target device

Spathy01, 2024-08-31:
I did all the steps but the payload.dd doesnt run when i plug in the pico. here are some screensthots of my files for my pico and payload: 
[image]
[image]

I have tried unplugging the pico and plugging it back into the host pc but still doesnt work. It only opens the picos file system as if it were a normal usb storage drive. 

Is there any way to fix this?

Spathy01, 2024-08-31:
i have realised that maybe its because it isnt being registered as a HID device but as a disk, could this be the issue even though i have all the neccessary files in my pico lib directory? if this is the issue how would i fix it.
[image]

dbisu, 2024-08-31:
Make sure you are using CircuitPython 8.x and not 9.x. The project hasn't been updated yet to support 9.x.

Spathy01, 2024-08-31:
> Make sure you are using CircuitPython 8.x and not 9.x. The project hasn't been updated yet to support 9.x.

I am using circuit python 8x but it still doesn't work?

dbisu, 2024-09-01:
Have you tried using one of the release bundles found [here](https://github.com/dbisu/pico-ducky/releases)?

Spathy01, 2024-09-01:
I solved the issue, must've been a problem on my part thanks tho.

dbisu, 2024-09-02:
Closing, resolved.

closed 2024-09-02 by dbisu