dbisu/pico-ducky #282: Adding croatian layout

Paolo631, 2024-09-17:
I need help adding the croatian layout to the ducky pi. thank you.

Paolo631, 2024-09-17:
@dbisu  help pls

dbisu, 2024-09-19:
Have you worked through the instructions [here](https://github.com/dbisu/pico-ducky?tab=readme-ov-file#changing-keyboard-layouts)?
If so, what errors are you getting?

Paolo631, 2024-09-19:
Folllowed the Instructions, copied Croatian Key Board layout to
adafruit_hid and changed the ducky code but It Just Isn't Working, When i
unplug It and plug It Back in its Just a masa storage device and i See All
the ducky Files in It but It Isn't executing scripts.

Paolo631, 2024-09-20:
> Make sure you are using circuitpython 8.x and not 9.x. The project files
> have not been updated to support 9.x yet.

I was using the 9x....

dbisu, 2024-09-21:
Did that resolve the issue?

Paolo631, 2024-09-25:
Sorry That i didnt reply, We had no Power for few Days. That did Not solve
the issue.

dbisu, 2024-09-26:
Please follow the steps in [DEBUG.md](https://github.com/dbisu/pico-ducky/blob/main/DEBUG.md) to get logs from the device.