dbisu/pico-ducky #283: Error, could not open file. (IMPORT command in payload)

g0lder, 2024-09-26:
Whenever I use the import command inside of the payload, it says, "Error, could not open file (x)"

simply create a payload, and try to import a module. I tried putting the module I needed in different directories, and it wouldn't work :(       (I was attempting to get adafruits_hid.mouse)

I expected to be able to import the file and use the module commands

(P.S. I am trying to get the mouse to move, I got the buttons to click by defining them in the duckyinpython.py file)

Crossshield45, 2024-10-11:
Put those files in the lib folder. The code doesn't use other folders as far as I know.

dbisu, 2024-12-14:
The import command imports other duckyscript files, not modules.
to include other python modules, you'll need to modify the python code.