dbisu/pico-ducky #292: Problem with USB enable/disable mode

TheMrM, 2024-11-06:
When I place jumpers on pin 18 (GND) and pin 20 (GPIO15), and plug it into my PC, the files are shown but the script also runs, which it should not.
These are all my files when the Py is running as CIRCUITPY
[image]

I don't know if this is the correct place to make an issue regarding the problem.
Also, this is the guide that I followed https://www.youtube.com/watch?v=ctCmOhoT9po

dbisu, 2024-11-07:
On a Pico W, you need the jumper wire to make it show up as a USB drive. Without the jumper it will not show up as a USB drive.
This sounds like it is running as designed.

TheMrM, 2024-11-07:
I am using the jump wires so I can see it as usb, but my problem is, it also runs the script.

dbisu, 2024-11-07:
Correct, you need to put a jumper from pin 1 to pin 3 to enter setup mode. In setup mode it will not auto run the scripts.

TheMrM, 2024-11-07:
I also did that, but then I don't have access to the files

TheMrM, 2024-11-07:
Wait you mean I need jumpers on both 1 and 3 , 18 and 20 , at the same time?

dbisu, 2024-11-07:
Yes, both at the same time. they are separate functions.

TheMrM, 2024-11-07:
OMG, I have not thought of that. Thank you very much! Will give it a go. And thanks for the time you took to help me out with the issue!

TheMrM, 2024-11-07:
Thank you very much! I just tried it out and it worked. Gonna close this issue now.

closed 2024-11-07 by TheMrM