dbisu/pico-ducky #294: Payload issue

Pat-Jane, 2024-11-24:
Everything works fine , from the files to the coding , the only issue is that when i plug the pico to any device it doesnt work . 
everything works like its supposed to but it just doesnt run the payload no matter how simple it is . 
I Tried disabling every antivirus , trying every keyboard layout . it just does not work. 
Thats about it , any help would be appreciated .

dbisu, 2024-11-25:
Please collect debug logs to help figure out what is happening.
Look [here](https://github.com/dbisu/pico-ducky/blob/main/DEBUG.md) for info on how to do that.

dbisu, 2024-11-26:
At least part of the problem is using circuitpython 9.x with the libraries currently in the repo. Either move back to 8.x or update the included libs to the 9.x versions.
I haven't had a chance to include support for 9.x. Hopefully I'll get some time in a month or so.

Pat-Jane, 2024-11-27:
hello @dbisu , ii ran the 8.x , and This is the PuTTY Logs : Auto-reload is on. Simply save files over USB to run them or enter REPL to disab                                                                                                                                                             le.

Press any key to enter the REPL. Use CTRL-D to reload.
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 14, in <module>
  File "duckyinpython.py", line 9, in <module>
ValueError: incompatible .mpy file

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 9.2.1 on 2024-11-20; Raspberry Pi Pico W with rp2040

closed 2024-11-28 by Pat-Jane