dbisu/pico-ducky #295: Not Working on Pico 2

marcus49007, 2024-12-07:
I encountered an issue where the script does not work on Pico 2. I am trying to implement the same as directed please help.

dbisu, 2024-12-08:
I would need more information on the errors you are getting. I haven't tried using on this board.

NF799, 2024-12-09:
Hi im also having the same issue, i completed every step but if i write a duckyscript and move it to the pico 2 W It just don't run. for other devices it's getting detected as a keyboard so that's good i think?

dbisu, 2024-12-10:
I believe the problem is likely mixing CircuitPython 9.x with the libraries that are from 8.x.
Try grabbing the 9x. libraries from [here](https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/tag/20241206) to match the installation of CircuitPython 9.x

johnfromul, 2024-12-10:
same problem, again downloaded "adafruit-circuitpython-bundle-9.x-mpy-20241210.zip" but without change...after pluged Pico2 in to USB port payload wasn´t injected

dbisu, 2024-12-10:
Copy these files from the bundle:

Navigate to `lib` in the recently extracted folder and copy `adafruit`_hid to the `lib` folder on your Raspberry Pi Pico.

Copy `adafruit_debouncer.mpy` and `adafruit_ticks.mpy` to the `lib` folder on your Raspberry Pi Pico.

Copy `asyncio` to the `lib` folder on your Pico.

Copy `adafruit_wsgi` to the `lib` folder on your Pico

johnfromul, 2024-12-11:
ok, my fail...i forgot change keyboard layout settings in "duckyinpython.py" from default:
**from keyboard_layout_win_fr import KeyboardLayout
from keycode_win_fr import Keycode**

to my choiced:
**from keyboard_layout_win_cz import KeyboardLayout
from keycode_win_cz import Keycode**

marcus49007, 2024-12-11:
@johnfromul  is it working for you ?
for me it is not working I have been trying since last 2 weeks 
Tested all approaches, tried circuitpython's 6.x to 9.x 
changed layout of keyboard as well 
I am in Asia and I am using English UK and English US keyboards 
If i am using English US keyboard layout do i need to explicitly change the keyboard layout setting in duckyinpython.py.
@dbisu Sir Please help 
I am first year in Diploma of cyber security course please help SIR

ToxicMonster, 2024-12-11:
The pico 2 uses a different controller chip `RP2350` rather than `RP2040` that the original pico uses. I don't believe the `RP2350` is supported by circuitpython yet.

marcus49007, 2024-12-11:
> I would need more information on the errors you are getting. I haven't tried using on this board.

What kind of information you need @dbisu  Sir, Would like to provide as much as information is required.  Please specify I am new in this field 
please help

johnfromul, 2024-12-11:
> @johnfromul is it working for you ? for me it is not working I have been trying since last 2 weeks Tested all approaches, tried circuitpython's 6.x to 9.x changed layout of keyboard as well I am in [...]

yes, it working with my Pico 2 (RP2350).....
Maybe you can try installer.bat or installer.sh, it is automated installation on your Pico device (this worked for me and after this i tryed it again with manual install)....but steps are same with manual installation.

marcus49007, 2024-12-11:
> The pico 2 uses a different controller chip `RP2350` rather than `RP2040` that the original pico uses. I don't believe the `RP2350` is supported by circuitpython yet.

@ToxicMonster I think it is supported 
could you please specify the source that claims this ?
as far as I gone through the reports and reviews it doesn't affect that much 

before even creating this issue #295 I researched this from side

marcus49007, 2024-12-11:
> keyboard_layout_win_cz

Do you use Czech Republic layout keyboard ?

johnfromul, 2024-12-11:
> > > The pico 2 uses a different controller chip `RP2350` rather than `RP2040` that the original pico uses. I don't believe the `RP2350` is supported by circuitpython yet.
> > 
> > 
> > @ToxicMonster I think it is supported could you please specify the source that claims this ? as far as I gone through the reports and reviews it doesn't affect that much
> > before even creating this issue #295 I researched this from side
> 
> I had just done a bit of research and the only information that I had found is that it supports the `2040` I may be wrong by this but I believe the issue will still be the different chip

it work on my Pico2 RP2350

johnfromul, 2024-12-11:
> > keyboard_layout_win_cz
> > @johnfromul
> > Do you use Czech Republic layout keyboard ?

yes

marcus49007, 2024-12-12:
`
if(board.board_id == 'raspberry_pi_pico'):
    # On Pi Pico, default to USB visible
    noStorage = not noStorageStatus
elif(board.board_id == 'raspberry_pi_pico_w'):
    # on Pi Pico W, default to USB hidden by default
    # so webapp can access storage
    noStorage = noStorageStatus
`
@dbisu  @johnfromul @ToxicMonster 
please help 
do i need to edit to raspberry_pi_pico_2 ?

marcus49007, 2024-12-12:
> yes, it working with my Pico 2 (RP2350)..... Maybe you can try installer.bat or installer.sh, it is automated installation on your Pico device (this worked for me and after this i tryed it again [...]

where are the installer.bat script ?

johnfromul, 2024-12-12:
> `if(board.board_id == 'raspberry_pi_pico'): # On Pi Pico, default to USB visible noStorage = not noStorageStatus elif(board.board_id == 'raspberry_pi_pico_w'): # on Pi Pico W, default to USB hidden [...]

no, i changed only setting of keyboard layout in "duckyinpython.py"

johnfromul, 2024-12-12:
> where are the installer.bat script ?

you need download wohle project to zip [https://gitlab.inria.fr/ddealmei/pico2-ducky/-/archive/main/pico2-ducky-main.zip](url) and inside is all include installer.bat and installer.sh

dbisu, 2024-12-14:
> `if(board.board_id == 'raspberry_pi_pico'): # On Pi Pico, default to USB visible noStorage = not noStorageStatus elif(board.board_id == 'raspberry_pi_pico_w'): # on Pi Pico W, default to USB hidden [...]

Yes, this will likely need support for the Pico 2 to be added.
I've ordered some Pico 2 boards. I should have some time in a couple of weeks to figure out the pico 2 support.

MishMush1, 2024-12-16:
Would the functionality of the pico 2 W be different to the pico W?

dbisu, 2024-12-17:
Functionally it should be similar. It will require a different build of circuitpython, which I don't think has been released yet. There are a couple of other code tweaks likely needed.
Not sure when I'll get some pico 2 w boards. I've just ordered some pico 2 boards.

MishMush1, 2024-12-20:
I have got a pico 2 W there is a circuit python avalible for it and ive put everything in. The bad usb part of it is working but I can't get it to host a webpage. I changed all the board id's

MishMush1, 2024-12-20:
Nevermind it works now

marcus49007, 2024-12-21:
> Functionally it should be similar. It will require a different build of circuitpython, which I don't think has been released yet. There are a couple of other code tweaks likely needed. Not sure when I'll get some pico 2 w boards. I've just ordered some pico 2 boards.

Thank you so much sir for effort!

dbisu, 2024-12-21:
I have Pico 2 boards now. I'll hopefully have time this next week to add support. It will probably be a bit before I get some Pico 2W boards.

MishMush1, 2024-12-21:
If you like I can give you the files for my pico 2 W board so you can add support. It works perfectly fine

dbisu, 2024-12-21:
If you just changed the code locations checking the board ids, I can probably make those changes. I'll create a branch and update this thread of you want test before I merge it.

MishMush1, 2024-12-21:
Yeah that's fine.

dbisu, 2024-12-21:
Thanks, I appreciate the help.

MishMush1, 2024-12-21:
No problem!

dbisu, 2024-12-21:
Branch has been created. [Pico2 support](https://github.com/dbisu/pico-ducky/tree/pico2_support)
I would appreciate if someone with a Pico 2W could test it. The code works on my Pico2.

MishMush1, 2024-12-21:
It works fine with Pico 2 W

dbisu, 2024-12-21:
Thank you for testing. I'll get this merged to in main shortly. I should have releases created later today.

closed 2024-12-21 by dbisu

marcus49007, 2024-12-26:
Great work dear @dbisu Sir. I submitted the synopsis on Ethical penetrating testing features offered  USB ducky (Pico2). My guide accepted with any doubt.
Thank you so much great work