dbisu/pico-ducky #296: Unable to save data to the dcuky

homelab192, 2024-12-11:
**Describe the issue**
I am just trying to run simple scripts like this:

REM 
GUI r
DELAY 1200
STRING powershell
ENTER
DELAY 500
STRING netsh wlan show profiles > D:\

**To Reproduce**
Fresh install,
Connect secrets,
create above payload file
windows test machine

**Expected behavior**
Windows says out-file : Access to the path 'D:\' is denied.

[image]

I am sure this is something simple that im over looking but im not sure which file to edit since doesnt have a cli i believe

dbisu, 2024-12-13:
Are you using a pico or pico w? which state do you have USB enable?

homelab192, 2024-12-25:
@dbisu  Thank you for taking the time to reply! I am using the pico w and have it so that the device shows up as usb. I did end up doing some digging on different repos. I was able to gather enough knowledge to accomplish the task another way. Which is having the ducky run a powershell script and then saving it to the ducky. Here is the saving code.

 # Function to find the drive named "DUCKY"
function Get-DuckyDrive {
    $allDrives = Get-WmiObject -Class Win32_LogicalDisk | Where-Object { $_.DriveType -eq 2 -or $_.DriveType -eq 3 }  # 2 = Removable, 3 = Fixed
    foreach ($drive in $allDrives) {
        if ($drive.VolumeName -eq "DUCKY") {
            return $drive.DeviceID
        }
    }
    return $null  # Return null if no drive named "DUCKY" is found
}

# Search for the "DUCKY" drive and save the system information
$duckyDrive = Get-DuckyDrive
if ($duckyDrive) {
    $TargetFilePath = "$duckyDrive\$FileName"
    $systemInfo | Out-File -FilePath $TargetFilePath
}

I understand this is not very clean nor reliable. I have a remote server that I pull the script from....I just havent spent the time to configure to send back. This was a proof of concept fix. I must have posted that like right away lol sorry.

dbisu, 2024-12-25:
Glad you found a solution.
Closing as resolved.

closed 2024-12-25 by dbisu