dbisu/pico-ducky #297: Duckyscript 3 Commands not Working

ToxicMonster, 2024-12-11:
**Describe the issue**
Duckyscript 3 commands are not working when added to `payload.dd` file including `WAIT_FOR_BUTTON_PRESS`, functions and variables

**To Reproduce**
Add one of the duckyscript 3 commands to payload

**Expected behavior**
Duckyscript 3 command executes as per [Hak5 docs](https://docs.hak5.org/hak5-usb-rubber-ducky/duckyscript-tm-quick-reference).

**Additional context**
Addition of duckyscript 3 commands would greatly increase functionality of pico as rubber ducky.

dbisu, 2024-12-14:
I'll have some time in a couple of weeks to take a look at this issue.

stilllearning65, 2026-01-29:
tried using the functions.dd as just a regular payload.dd file.
nothing happens.

Auto-reload is off.
code.py output:
starting monitor_buttons
progStatus False
Finding payload
starting monitor_led_changes
starting blink_pico_led
Running
runScript: REM Example Function

runScript: FUNCTION COUNTDOWN()

runScript: COUNTDOWN()

runScript:
-- MANUALLY TERMINATED AFTER 1 MIN. --
Traceback (most recent call last):
  File "code.py", line 91, in <module>
  File "asyncio/core.py", line 346, in run
  File "asyncio/core.py", line 283, in run_until_complete
  File "duckyinpython.py", line 599, in monitor_buttons
  File "adafruit_debouncer.py", line 100, in update
KeyboardInterrupt:

Code done running.