dbisu/pico-ducky #298: ducky isnt workuing in rpi pico 1

Linkinmyyt, 2024-12-20:
i held BOOTSEL when pluging it in but still nothing, i dont think i did anything wrong, i mean when i copied it over i didnt select them all at once, in the instructions there were 4 things to install in one step, idk if it helps but im using this micro usb cable
[image]
[image]
[image]

dbisu, 2024-12-21:
What version of CircuitPython did you install? I'm working on adding support for 9.x soon.

dbisu, 2024-12-21:
Try updating to the v3.0 release to see if that fixes your problem.

Linkinmyyt, 2024-12-22:
can you give me a link bc im not finding it

dbisu, 2024-12-22:
You can find the release bundles [here](https://github.com/dbisu/pico-ducky/releases/tag/v3.0)

Linkinmyyt, 2024-12-22:
still not working, i did everything in the instructions besides setup mode and i downloaded the right uf2 file too.

Linkinmyyt, 2024-12-22:
i know the cable isn't the one at fault bc i could upload stuff with it.

Linkinmyyt, 2024-12-22:
it registers as a hid device bc when i unplug it these two disappear
[image]

dbisu, 2024-12-22:
Follow the instructions [here](https://github.com/dbisu/pico-ducky/blob/main/DEBUG.md) to collect more debug data.

Linkinmyyt, 2024-12-23:
It was COM5 for me but what do i do now
[image]

Linkinmyyt, 2024-12-23:
It was COM5, but i got this screen so i left it for half an hour and it was still the same
[image]

dbisu, 2024-12-23:
In the putty window, hit Ctl+C, then Ctl+D and then send the debug output.

Linkinmyyt, 2024-12-23:
heres the output
[image]

dbisu, 2024-12-23:
Looks like the file payload.dd is missing or unreadable. double check that file is there and has text contents.

Linkinmyyt, 2024-12-23:
IT WORKED! i just had to make it with notepad++, thanks for the help

closed 2024-12-23 by Linkinmyyt