dbisu/pico-ducky #301: Payload not executing with alternative (UK) keyboard layout

leele2, 2024-12-29:
**Describe the bug**
When using the **UK.zip, I could not get the payload to run, but had no issues once I switched to default (us).zip.

**Expected behavior**
Expected payload to execute

**Debug info**
It seems (atleast in the UK) language .zip the duckyinpython.py has the following on line 23

`from keyboard_layout_win_uk import KeyboardLayout`

I believe this to be incorrect and should instead say

`from keyboard_layout_win_uk import KeyboardLayout as KeyboardLayout`

**Additional context**
Seems the typo exists in the main repo, so possibly the bug exists in all languages.

dbisu, 2024-12-29:
hmm. ok. I can make that change.
Thanks for letting me know.

closed 2024-12-29 by dbisu

Lucaaa1234, 2025-01-12:
it still doesn't work for me, even after the patch.

**Describe the bug**
When using the **IT.zip, I cannot get the payload to run.
If I use the default .zip, which is the US one, it works, but in this way if my Payload contains something like "1+1=2" it actually writes "1^1ì2", which is very annoying and for this motive I wanted to use the IT Language.

I also tried to follow the **Full Install Instructions** and after it to manually change the Keyboard Layout following **Changing Keyboard Layouts**, but the result didn't change.