dbisu/pico-ducky #307: not working with win da layout

lucasrj, 2025-01-11:
**Describe the bug**
i was trying to use release 3.1 with win da layout but it did not do anything

i was using pico 2

**To Reproduce**
Steps to reproduce the behavior:
1. follow Quick Start Guide

**Expected behavior**
device mounts as storage, hid and serial but nothing more

**Debug info**
i opened there serial and presed ctrl+d to reload the program
serial output

```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 9.2.1 on 2024-11-20; Raspberry Pi Pico 2 with rp2350a
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 14, in <module>
  File "duckyinpython.py", line 163, in <module>
NameError: name 'Keyboard' is not defined

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```

**Additional context**
i uncommned line 15 in duckinpython.py and then it seamed to work

dbisu, 2025-01-12:
Ah, yes, that appears to be a bug in the creation script. I'll get that fixed.

meraklimaymun, 2025-01-12:
> Ah, yes, that appears to be a bug in the creation script. I'll get that fixed.

Unfortunately, I don't get the proposed solution and it doesn't seem to effect anything. Line 15 in duckinpython.py is "import time". How can import time change the wrong keyboard layout?

dbisu, 2025-01-12:
The line in the current code is line 25.
The fix is to ensure this line is not commented out
from adafruit_hid.keyboard import Keyboard

meraklimaymun, 2025-01-12:
> The line in the current code is line 25. The fix is to ensure this line is not commented out from adafruit_hid.keyboard import Keyboard

Yes, but line 25 is already not commented out.

closed 2025-01-16 by dbisu