dbisu/pico-ducky #308: it still doesn't work for me, even after the patch.

Lucaaa1234, 2025-01-12:
it still doesn't work for me, even after the patch.

**Describe the bug**
When using the **IT.zip, I cannot get the payload to run.
If I use the default .zip, which is the US one, it works, but in this way if my Payload contains something like "1+1=2" it actually writes "1^1ì2", which is very annoying and for this motive I wanted to use the IT Language.

I also tried to follow the **Full Install Instructions** and after it to manually change the Keyboard Layout following **Changing Keyboard Layouts**, but the result didn't change.

_Originally posted by @Lucaaa1234 in https://github.com/dbisu/pico-ducky/issues/301#issuecomment-2585785537_

dbisu, 2025-01-13:
I don't have other keyboard layouts. Could you try getting debug logs to see if there are errors.
Another thing to try is to try to capture the keycodes that are being sent and compare with the expected keycodes.

Lucaaa1234, 2025-01-13:
> Another thing to try is to try to capture the keycodes that are being sent and compare with the expected keycodes.

I did it this way: first, I checked how the keys would be translated according to the two different layouts. As the first step, the payload changes the system layout from IT to US, allowing the rest of the payload to be interpreted literally.
The last thing I do before ending the script is to reverse the operation, switching back from US to IT.
For now, this solution allows me, with just one line in the payload, to avoid going crazy manually translating all the special characters.
Thank you for your response.

closed 2025-01-13 by Lucaaa1234