dbisu/pico-ducky #309: Turkish key layout doesn't work

meraklimaymun, 2025-01-12:
I followed the guide to the last detail, copied files, uncommented and the code etc... etc...  but the payload doesn't work when Turkish language is set. When I set it in English, it works but presses wrong keys.

The only way to make it work properly is changing Windows' key layout to English which defeats the whole purpose of pico-ducky.

dbisu, 2025-01-12:
There is a known problem with the 3.1 release. I'm working on a fix. Will probably be a couple of days. See issue #307 for a quick fix.

dbisu, 2025-01-13:
Can you include information from the debug serial? I don't have other keyboard layouts to use for testing.

meraklimaymun, 2025-01-13:
> Can you include information from the debug serial? I don't have other keyboard layouts to use for testing.

I would love to but I have no idea how to do that. I'm newbie programmer. Firstly, you mean running the code in an IDE and copying the whatever text comes up in the "shell" section right? I have Thonny as IDE and I set it up with Circuit Python interpreter. Would that work? If so, which .py file should I run? If it's "ducyinpython.py" then here's the text:

`Traceback (most recent call last):
  File "<stdin>", line 50, in <module>
AttributeError: type object 'Keycode' has no attribute 'POWER'`

dbisu, 2025-01-14:
OK, that is super helpful. Looks like one of the recent PRs broke some non-US keyboards. I'll have to do some research on a fix.

meraklimaymun, 2025-01-21:
I can confirm this issue still exists. Turkish keyboard does not work. This time I didn't get any errors in Thonny but the payload still didn't run.

dbisu, 2025-01-22:
Can you share the debug out put from Thonny while it boots and runs?

dbisu, 2025-03-23:
Closing, no updates.
If you get new debug information, please open a new issue.

closed 2025-03-23 by dbisu