dbisu/pico-ducky #312: duckyinpython code refactor

TzurSoffer, 2025-01-16:
The `duckyinpython.py` is getting quite large and, in my opinion, messy. How would you feel about a refactor to an object-oriented structure? I am happy to submit a PR doing so. It would make adding features significantly easier for you and other contributors and make the code neater. It would look something similar to my [DuckyScript-3.0-to-c converter](https://github.com/TzurSoffer/DuckyScript-3.0-to-c/blob/main/convertor.py).

dbisu, 2025-01-16:
While I appreciate the offer, I've got some other things I'd like to tackle first. It might be something to look at in a few months.

TzurSoffer, 2025-01-16:
I can do it if you are busy, I just want to know if it's a change you would like me to make, if it's something you would like to do yourself, or if you like the code the way it is. It's important to me since I am trying to add while loops and conditional statements, and in the code's current form, it's not something straightforward to do.

dbisu, 2025-01-17:
There are some other changes I'd like get done first. There are some older PRs that I've been putting off reviewing. There are also some issues with the web server and delays in scripts that I need to debug.

TzurSoffer, 2025-01-17:
I understand. Thank you for your feedback.

closed 2025-01-17 by TzurSoffer

dbisu, 2025-01-17:
I do appreciate the code changes you've been making.