dbisu/pico-ducky #320: Picoducky payloads not working on spanish computers even changing the keyboard language

hadfff, 2025-01-22:
**Describe the issue**
A clear and concise description of the symptoms

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Debug info**
If possible, include debug serial data.
On Windows, use PuTTY, connect to the debug serial port (commonly COM3, but could vary)

On Linux, use minicom (minicom -b 115200 -o -D /dev/ttyACM0, where ttyACM0 corresponds to your device)

**Additional context**
Add any other context about the problem here.

dbisu, 2025-01-24:
If you are using the v3.1 release, there is a known issue in it. You need to uncomment line 15 of duckyinpython.py
Hopefully by this weekend I'll get a new release out that has the fix. in the meantime, you can just edit that file directly.

hadfff, 2025-01-27:
Thanks man ill try it when i get home

dbisu, 2025-03-23:
I've created release v3.2 that should have the fix for this issue. Please retest after updating.