dbisu/pico-ducky #322: Add support of 2040 zero

djpostka, 2025-01-30:
Hi,

would it be possible to add support for rp2040-ZERO? Thanks

dbisu, 2025-02-01:
Might be possible. I don't have one of those boards to use for testing. I would need some requirements for GPIOs to use for setup mode and USB hide/show.

djpostka, 2025-02-01:
Hi,

One guys was made an adaptation but is not up to date: https://github.com/sufigueroa/pico-ducky-RP2040-Zero

For the stealth mode, he use the pin GP15 and GND.

djpostka, 2025-02-01:
I have one rp2040-zero, i can make test if necessary.

TzurSoffer, 2025-03-03:
I don't know if this helps, but if you use the https://github.com/sufigueroa/pico-ducky-RP2040-Zero and copy the duckyinpython.py from this repo over to the adaptation, it should have up-to-date payload handling.

AvadaKedavraEagle, 2025-06-19:
I used Google Gemini to create a port of pico ducky to the 2040 zero and has multiple payload execution, setup mode, and stealth mode. It is largely untested but executed scripts in my testing.

[RP2040 python pico ducky scripts.zip](https://github.com/user-attachments/files/20824054/RP2040.python.pico.ducky.scripts.zip)