dbisu/pico-ducky #323: Payload not executing pico W

JumpingDonut, 2025-02-09:
**Describe the issue**
Payload not executing

**Screenshots**

[image]

[image]

[image]

ProgramingPro2, 2025-02-13:
Bump, same issue

dbisu, 2025-02-13:
Please follow the instructions [here](https://github.com/dbisu/pico-ducky/blob/main/DEBUG.md) to post the logs from the debug terminal.
If you are running one of the release bundles, they are currently broken. I'm hoping to get a new release out soon.

JumpingDonut, 2025-02-19:
> Please follow the instructions [here](https://github.com/dbisu/pico-ducky/blob/main/DEBUG.md) to post the logs from the debug terminal. If you are running one of the release bundles, they are currently broken. I'm hoping to get a new release out soon.

Hey, thanks for reaching out, im not at home at the moment. When I come back (1 week or so) ill post the logs.

krobol, 2025-03-23:
Got it to work yesterday. The log says something like "Keyboard is not defined". "Keyboard" is defined in line 15 of  duckingpython.py and for some reason that line is commented out. Just remove the # in front of the line and it should work

dbisu, 2025-03-23:
If you are using an non-US keyboard layout, there is an issue in the v3.1 release bundles. I haven't had a chance to create an updated release.

dbisu, 2025-03-23:
I've created release v3.2 that should have the fixes for the keyboard layout issues.

vzxid, 2025-06-05:
I have the same issue. My duckingpython.py is correct. Anyone have any fixes?