dbisu/pico-ducky #332: Payload not loading

Les-Nouveaux-Devs, 2025-05-31:
**Describe the issue**
I added the uf2 file to my pico, it did its thing then I replaced the code.py with the duckything.py (I did rename it to code.py), I added the hid library to lib, I added the 2 keyboard layouts to the pico lib folder and I did the necessary changes to the code.py

**To Reproduce**
I wrote a very simple payload, renamed it payload.dd, unpluged then pluged again and nothing happened

**Expected behavior**
I expected to see a notepad window but nothing happened

[CIRCUITPY.zip](https://github.com/user-attachments/files/20535421/CIRCUITPY.zip)

dbisu, 2025-05-31:
Don't rename duckyinpython.py to code.py. There is already a code.py file.
Try loading one of release packages if you want a ready to go version.
Otherwise, make sure you are following the installation instructions and getting all of the library files placed correctly.

Les-Nouveaux-Devs, 2025-05-31:
Ok so I just installed the french release in 5 mins and it worked, so thankyou for answering so fast or I would have lost even more hours into this, also you can't find adafruit_wsgi anywhere unless you google it specifically or from a release.